OpenDevicePartnership/embassy-imxrt #338: Need to move I2C State Machine Handler to ISR for Maintaining Performance

JamesHuard, 2025-03-06:
Due to the potential for delays between async calls, the state machine transitions (i.e., START->WRITE->CONT->READ->STOP) should be handled in the ISR directly, so that the async i2c transaction function simply polls a future for when the entire transfer is complete, instead of each individual state transition being polled. This will help guarantee I2C bus performance regardless of what work the CPU has bound

felipebalbi, 2025-05-22:
> Due to the potential for delays between async calls, the state machine transitions (i.e., START->WRITE->CONT->READ->STOP) should be handled in the ISR directly, so that the async i2c transaction [...]

Would you have some performance numbers related to this issue?

JamesHuard, 2025-05-29:
> Would you have some performance numbers related to this issue?

The motivations and performance data gathered as part of this embassy feature are a clear demonstration of this issue: https://github.com/embassy-rs/embassy/pull/4035