OpenDevicePartnership/embassy-imxrt #354: i.MXRT1011 support and RAL

i509VCB, 2025-03-24:
I came across this project and was almost going to start an embassy port for the imxrt1011 independently.

I did notice this project is based off the SVDs and isn't using imxrt-ral. Is there some reason for that choice? I would suspect that the ral would be a little more internal consistent (I know SVDs can be buggy and inconsistent from experience).

felipebalbi, 2025-03-24:
> I came across this project and was almost going to start an embassy port for the imxrt1011 independently.
>
> I did notice this project is based off the SVDs and isn't using imxrt-ral. Is there some [...]

Hi @i509VCB,

We should really have posted this HAL for embassy inclusion already. Anyway, I didn't know about the RAL and when I started, I sort of followed what MIMXRT500 was doing. I guess it would not be too much work to replace the PACs with the RAL and update the drivers accordingly. Let me spend some time looking at this. At a minimum, would have to add eSPI support for the RAL, I guess.

i509VCB, 2025-03-24:
Regarding upstreaming to embassy, there is a new NXP hal. (Although it only contains very incomplete LPC55 support). https://github.com/embassy-rs/embassy/tree/main/embassy-nxp

An alternative of course is to use the svds with the 600/500 and then use the RAL for the rest of the imxrt parts.

i509VCB, 2025-03-24:
Yes that sounds reasonable.

From what I recall in embassy-nxp, there probably needs to be a pull request first to move a few things around so that the lpc55 stuff isn't overlapping (currently sits in root directory of src in the crate). I can try to help with that and also on the 1011 support.

felipebalbi, 2025-03-24:
> Yes that sounds reasonable.
> 
> From what I recall in embassy-nxp, there probably needs to be a pull request first to move a few things around so that the lpc55 stuff isn't overlapping (currently sits in root directory of src in the crate). I can try to help with that and also on the 1011 support.

Sounds good to me, let me have a quick look on what is there already. We can move chip-specific details to a submodule `chips` that gets selected based on feature flags, much we like I did for $this HAL.

i509VCB, 2025-04-01:
Related, someone I was talking to already has part of a i.mx1xxx HAL written:
https://github.com/korbin/embassy-imxrt
https://github.com/korbin/embassy-boot-imxrt
https://github.com/korbin/imxrt-pacs/
https://github.com/korbin/chiptool/tree/delete-duplicate-enum-fields

Also I've looked more into i.mxrt-ral and it feels... awkward in some ways to use. I'm not a huge fan of the macro register access syntax.

I'd probably work from the stuff korbin donated to public and do something based on chiptool, but deduplicate common peripherals. I'm around in the embassy matrix chat (same username as here) if you wanted to discuss this some more over there in a more *real time* fashion: https://matrix.to/#/#embassy-rs:matrix.org

felipebalbi, 2025-04-03:
PR for embassy-imxrt pending upstream (https://github.com/embassy-rs/embassy/pull/4042). Thanks @i509VCB for the help reviewing ;)

closed 2025-04-03 by felipebalbi

teburd, 2025-04-03:
RAL could support imxrt500/600 fairly easily, there's some stuff there that merges/reduces duplicate blocks in the nxp svds and such.

Intially the RAL was done the way it was because svd2rust would generate a 1mloc crate that took 10min to build. Maybe this isn't true anymore, but the ral approach is effectively "instant" compiles (macros that look at a module hierarchy).

i509VCB, 2025-04-03:
Yeah the https://github.com/imxrt-rs/imxrt-boot-gen stuff looks interesting and is largely independent of the rest of the stuff.

felipebalbi, 2025-04-03:
> I would however look at the way imxrt-rs does its configuration block and runtime bits, would be cool to see collaboration there.

If there are opportunities to share code, all the better. Currently we're relying on [mimxrt600-fcb](https://docs.rs/mimxrt600-fcb/latest/mimxrt600_fcb/) crate. I'm more than happy to deprecate that crate, if there's a better option.